danieleff/STM32GENERIC #12: Nucleo F030R8 Blink not compiling

ChrisMicro, 2017-05-15:
First error:
 extern PCD_HandleTypeDef hpcd_USB_FS;
..

danieleff, 2017-05-22:
This should now be fixed in the master branch.

ChrisMicro, 2017-05-23:
Compiling and programming with ST-Link now working.
Trying to download  by "Mass Storage" results in:
" NODE_F030R8 not found. Please ensure the device is correctly connected "

danieleff, 2017-05-23:
If you plug in the board, what is the name of the new drive?

ChrisMicro, 2017-05-23:
NUCLEO

danieleff, 2017-05-26:
So drive names for boards:
- Nucleo F411RE: NODE_F411RE
- Nucleo F030R8: NUCLEO
- Discovery L0: DISCOVERY
- Discovery F7: DIS_F746NG

Great... would be good to have a list of all of these.

ChrisMicro, 2017-06-01:
Hi Daniel,
I just realized that you solved the issue. Great work! 
Thanks,
Chris

closed 2017-06-01 by ChrisMicro