danieleff/STM32GENERIC #14: Whetstone benchmarks to slow

ChrisMicro, 2017-05-23:
STM32F103,72MHz is ok:
Loops: 1000, Iterations: 1, Duration: 13100 ms.
C Converted Single Precision Whetstones: 7.63 MIPS

Nucleo F030R8, 48Mhz?
Loops: 1000, Iterations: 1, Duration: 46423 ms.
C Converted Single Precision Whetstones: 2.15 MIPS

ChrisMicro, 2017-05-23:
STM32F7 Discovery
Loops: 1000, Iterations: 1, Duration: 2150 ms.
C Converted Single Precision Whetstones: 46.51 MIPS

The STM32F4 Discovery is about 109MIPS and the F7 should be faster.

danieleff, 2017-06-06:
With the hardware float update, STM32F7 Discovery should be now much faster.
I do not know about F030R8, it might be just that slow.

ChrisMicro, 2017-06-06:
Hi Daniel, now the F7 is really faster:
Loops: 1000, Iterations: 1, Duration: 493 ms.
C Converted Single Precision Whetstones: 202.84 MIPS
very good :+1: 

Regarding the F030@48MHz: I would expect that it is faster than the 2.15 MIPS because the F103 has 7.63@72Mhz. Could it be a waitstate issue ?