danieleff/STM32GENERIC #15: BluePill ILI9341 not working with certain pins

ChrisMicro, 2017-06-02:
I have connected an ILI9341 display to a BluePill STM32F103.
It is working with the following configuration:

#define TFT_DC   PA2
#define TFT_RST  PA3
#define TFT_CS   PA4

#define TFT_CLK  PA5
#define TFT_MISO PA6
#define TFT_MOSI PA7

If I try to use PB3, PB4, PA14 it is not working.

My guess: Probably JTAG is enabled on the BluePill and it blocks the pins.

danieleff, 2017-06-18:
I added menu item, where it is possible to select if you want to enable JTAG/SWD. 
Can you try it?

ChrisMicro, 2017-06-18:
Confirmed, I tested all this pins
PA13, PA14, PA15, PB3, PB4,PA11,PA12
All are working when using the drop down menu to disable alternate functions.
Good Work, Daniel.
Thanx

closed 2017-06-18 by ChrisMicro