danieleff/STM32GENERIC #16: STM32F7 Discovery tft.setRotation not working

ChrisMicro, 2017-06-03:


danieleff, 2017-06-03:
Thanks for testing.
First I would like somebody with a STM32F429ZI Discovery board to implement it for that display, should be 
similar to LTDC_F746_Discovery.h, but needs to check the cubemx examples.
After that, I will implement rotation.

ChrisMicro, 2017-06-03:
> Thanks for testing.
Welcome :-) 
Just one question: could you turn the display preliminary by 180° as default  ? 
( only if it is possible without effort ... ) 
Because the USB cable points in my direction and it would be much better from the opposite site.

danieleff, 2017-06-03:
Change the `buffer[x + y * W] = color` to `buffer[(W - x - 1) + (H - y - 1) * W] = color` in
https://github.com/danieleff/STM32GENERIC/blob/master/STM32/libraries/LTDC/src/LTDC.h#L164

It has to be changed in all 4 methods below too, not just 1. Thats basically 180.

Pretty cool that writing to the buffer changes the pixel on the screen.

ChrisMicro, 2017-06-03:
>Pretty cool that writing to the buffer changes the pixel on the screen.
Yes, I like it :-)
I've just made more experiments with a my new voltmeter example
https://github.com/ChrisMicro/STM32GENERIC/blob/master/STM32/libraries/BoardExamples/examples/Discovery746NG/TFT_voltmeter/TFT_voltmeter.ino

- The cursor positioning seems not to  work ( line 66 )
- The line wrap seems to be to short ( at 2/3 of the screen it wraps )
- float shows only 2 post decimal positions

Just an observation:
I made measure the voltage with a 100KOhm resistor in front of the ADC for protection.
If the ADC is running, a voltage drop of 300mV occurs which means it draws 3uA in active state.

danieleff, 2017-06-03:
Cursor positioning is in pixels, so something like `tft.setCursor(50, 100);`
comment out the `tft.setRotation(1);` because that would be portrait mode. wide screen is 0 and 2
add precision parameter to print like `tft.print(voltage, 6);`

ChrisMicro, 2017-06-03:
Thanks. I changed it in the repository.
I added a simple oscilloscope with auto trigger:
https://github.com/ChrisMicro/STM32GENERIC/blob/master/STM32/libraries/BoardExamples/examples/Discovery746NG/TFT_oscilloscope/TFT_oscilloscope.ino

ChrisMicro, 2017-06-03:
I've just added a mandelbrot set example:
https://github.com/ChrisMicro/STM32GENERIC/tree/master/STM32/libraries/BoardExamples/examples/Discovery746NG/TFT_mandelbrot
Well ... double buffering wouldn't be bad ...