danieleff/STM32GENERIC #22: Spi compile error for stm32f411re

thetazzbot, 2017-06-06:
I've forked the code and intend to fix this because I need to try to use it.

I am running a Nucleo STM32F411RE board. I open the Example SPISelfTest sketch and compile it fine under the BluePill board selection. When attempting to use the F4, it fails with a compile error. The reason for the error I believe is that the STM32F411RE has 5 SPI ports and the SPI.h file only enumerates three of them when building out the SPIx_ChannelTX and SPIx_ChannelRX defines.

`
C:\Program Files (x86)\Arduino\hardware\STM32GENERIC\STM32\libraries\SPI\src\SPI.cpp:81:17: error: 'SPI4_ChannelTX' was not declared in this scope

_ChannelTX = SPI4_ChannelTX;

             ^
C:\Program Files (x86)\Arduino\hardware\STM32GENERIC\STM32\libraries\SPI\src\SPI.cpp:82:17: error: 'SPI4_ChannelRX' was not declared in this scope

_ChannelRX = SPI4_ChannelRX;
C:\Program Files (x86)\Arduino\hardware\STM32GENERIC\STM32\libraries\SPI\src\SPI.h:114:22: note: in expansion of macro '_SPIx_DMA'

#define SPIx_DMA(a) _SPIx_DMA(a)

`

I believe the source of the problem is in this list of defines

#define SPI3_StreamTX 1_Stream5
#define SPI3_StreamRX 1_Stream0
#define SPI3_ChannelTX DMA_CHANNEL_0
#define SPI3_ChannelRX DMA_CHANNEL_0

It stops at 3. It should probably follow the same paradigm as some other code wherein:
#ifdef SPI4 #define SPI4_StreamTX 1_Stream5 #define SPI4_StreamRX 1_Stream0 #define SPI4_ChannelTX DMA_CHANNEL_0 #define SPI4_ChannelRX DMA_CHANNEL_0 #endif #ifdef SPI5 #define SPI5_StreamTX 1_Stream5 #define SPI5_StreamRX 1_Stream0 #define SPI5_ChannelTX DMA_CHANNEL_0 #define SPI5_ChannelRX DMA_CHANNEL_0 #endif

Thoughts?

Cheers,
Mark

danieleff, 2017-06-06:
This is the result of unfinished DMA, simply uncommenting those few lines (SPI3/4/5) will make it work. (Unless you need SPI4/5/6)

danieleff, 2017-06-06:
I fixed the compilation error. Yes those defines should continue for 4/5/6, currently they are missing.

thetazzbot, 2017-06-06:
Thanks Daniel,  you beat me too it.

closed 2017-06-18 by danieleff