danieleff/STM32GENERIC #24: timer library

cnc4less, 2017-06-18:
will be perfect to have a timer library like the lib-maple one also to run https://github.com/bubulindo/STM32Encoder
also have full control on the timers in the chip

danieleff, 2017-06-19:
On one hand, a simple "I want to run this code every x microseconds" timer would be good. 
[IntervalTimer ](https://www.pjrc.com/teensy/td_timing_IntervalTimer.html)API is very nice for that.  Or even just `scheduleRepeated(microseconds, callback)`. Much simpler than trying to do the same with libmaple HardwareTimer.

On the other hand, a class like HardwareTimer to access more functionality of the chip can be done too.

ChrisMicro, 2017-06-20:
A very good starting point could be the [timer HAL example](https://github.com/danieleff/STM32GENERIC/tree/master/STM32/libraries/BoardExamples/examples/Discovery407VG/HAL) which could be directly used for that purpose.