danieleff/STM32GENERIC #29: Error when compiling.

Lennyz1988, 2017-07-25:
Arduino: 1.8.
Windows 10 x64
Running the latest version of this library.
I'm just picking a random sketch from this library, but it happens to all the sketches I tried.

```
In file included from C:\Users\Lennart\Documents\Arduino\hardware\STM32GENERIC-master\STM32\cores\arduino/Stream.h:26:0,

                 from C:\Users\Lennart\Documents\Arduino\hardware\STM32GENERIC-master\STM32\cores\arduino\stm32\SerialUART.h:28,

                 from C:\Users\Lennart\Documents\Arduino\hardware\STM32GENERIC-master\STM32\cores\arduino\stm32\SerialUART.cpp:29:

C:\Users\Lennart\Documents\Arduino\hardware\STM32GENERIC-master\STM32\cores\arduino/Print.h:94:39: error: 'va_list' has not been declared

     size_t printf(const char* format, va_list);

                                       ^

exit status 1
Error compiling for board BLACK F407VE/ZE/ZG boards.

```
What am I doing wrong?

Sketch:

```
void setup()
{
  pinMode(LED0_BUILTIN, OUTPUT);
  pinMode(LED1_BUILTIN, OUTPUT);
  
  digitalWrite(LED0_BUILTIN, HIGH); // led off, led is active low
  digitalWrite(LED1_BUILTIN, HIGH); // led off, led is active low
}

void loop()
{
    digitalWrite(LED0_BUILTIN, LOW); // led on
    delay(1000);
    digitalWrite(LED0_BUILTIN, HIGH);// led off
    
    digitalWrite(LED1_BUILTIN, LOW); // led on
    delay(1000);
    digitalWrite(LED1_BUILTIN, HIGH); // led off
}
```

danieleff, 2017-07-26:
Thanks.

I uploaded a fix, can you try now? 

This error comes up when compiling with arm-gcc 4.8, while I mainly test with arm-gcc 6.1

Lennyz1988, 2017-07-26:
Thank you! Now it compiles perfectly!

closed 2017-07-26 by Lennyz1988