danieleff/STM32GENERIC #35: STM32F030 series test on Arduino IDE

csnol, 2017-08-25:
Hi, All.
I'm testing STM32F030 Series F030F4 \ F030K6 \ Nucleo64 F030R8
SPI / Serial / PWM / digitalRead and Write function OK. 
But ADC input always got 0 on PA0-PA7;  (Use Arduino IDE example: AnalogInOutSerial）
Below information  gotten from serialport: and displayed 0V on multimeter too.
>sensor = 0	 output = 0
>sensor = 0	 output = 0
>sensor = 0	 output = 0
>sensor = 0	 output = 0
>sensor = 0	 output = 0

What have I missed?

appreciate for any helps.

BR.

C.S Nol

csnol, 2017-08-25:
The codes , there is only 2ms delay in the sketch. but I got more than 500ms

``` c++ 
const int analogInPin = A0;  // Analog input pin that the potentiometer is attached to
const int analogOutPin = A1; // Analog output pin that the LED is attached to

int sensorValue = 0;        // value read from the pot
//int outputValue = 0;        // value output to the PWM (analog out)

void setup() {
  // initialize serial communications at 9600 bps:
  Serial.begin(115200);
}

void loop() {
  // read the analog in value:
  sensorValue = analogRead(analogInPin);
  // map it to the range of the analog out:
  //outputValue = map(sensorValue, 0, 1023, 0, 255);
  // change the analog out value:
  analogWrite(analogOutPin, sensorValue/4);

  // print the results to the serial monitor:
  Serial.print("sensor = ");
  Serial.print(sensorValue);
  Serial.print("\t output = ");
  Serial.println(sensorValue/4);

  // wait 2 milliseconds before the next loop
  // for the analog-to-digital converter to settle
  // after the last reading:
  delay(2);
}
```

danieleff, 2017-08-26:
Thank you for reporting. 
I get the same problem with L0, so analog is different on these chips.

danieleff, 2017-08-27:
This should be fixed in 0531a2eba1ca50f5823b344ca16f0ee6bbfd45c9, could you try again?

csnol, 2017-08-27:
I've to move a lots of codes from CubeMX to settled this problem. 
Useful but terrible !

see below:
``` c++ 

#define Serialx SerialUART1

ADC_HandleTypeDef hadc;
DMA_HandleTypeDef hdma_adc;

/* Variable used to get converted value */
__IO uint16_t uhADCxConvertedValue = 0;
ADC_ChannelConfTypeDef sConfig;
// the setup routine runs once when you press reset:
void setup() {
  MX_ADC_Init();
  // initialize serial communication at 9600 bits per second:
  Serialx.begin(115200);
  Serialx.println("Hello CSNOL!");
  delay(1000);
  
}

// the loop routine runs over and over again forever:
void loop() {
  // read the input on analog pin 0:
  if(HAL_ADC_Start(&hadc) != HAL_OK)
  {
    /* Start Conversation Error */
    while(1);
  }
   HAL_ADC_PollForConversion(&hadc, 10);
  /* Check if the continuous conversion of regular channel is finished */
  if((HAL_ADC_GetState(&hadc) & HAL_ADC_STATE_EOC_REG) == HAL_ADC_STATE_EOC_REG)
  {
    /*##-5- Get the converted value of regular channel  ######################*/
    uhADCxConvertedValue = HAL_ADC_GetValue(&hadc);
  }
  // print out the value you read:
  Serialx.print(" A1 =  ");
  Serialx.println(uhADCxConvertedValue);
  delay(100);        // delay in between reads for stability
}

static void MX_ADC_Init(void)
{
  ADC_ChannelConfTypeDef sConfig;

    /**Configure the global features of the ADC (Clock, Resolution, Data Alignment and number of conversion) 
    */
  hadc.Instance = ADC1;
  hadc.Init.ClockPrescaler = ADC_CLOCK_SYNC_PCLK_DIV4;
  hadc.Init.Resolution = ADC_RESOLUTION_12B;
  hadc.Init.DataAlign = ADC_DATAALIGN_RIGHT;
  hadc.Init.ScanConvMode = ADC_SCAN_DIRECTION_FORWARD;
  hadc.Init.EOCSelection = ADC_EOC_SINGLE_CONV;
  hadc.Init.LowPowerAutoWait = DISABLE;
  hadc.Init.LowPowerAutoPowerOff = DISABLE;
  hadc.Init.ContinuousConvMode = DISABLE;
  hadc.Init.DiscontinuousConvMode = DISABLE;
  hadc.Init.ExternalTrigConv = ADC_SOFTWARE_START;
  hadc.Init.ExternalTrigConvEdge = ADC_EXTERNALTRIGCONVEDGE_NONE;
  hadc.Init.DMAContinuousRequests = DISABLE;
  hadc.Init.Overrun = ADC_OVR_DATA_OVERWRITTEN;
  if (HAL_ADC_Init(&hadc) != HAL_OK)
  {
   while(1);
  }

    /**Configure for the selected ADC regular channel to be converted. 
    */
  //sConfig.Channel = ADC_CHANNEL_0;
  sConfig.Rank = ADC_RANK_CHANNEL_NUMBER;
  sConfig.SamplingTime = ADC_SAMPLETIME_28CYCLES_5;
  //if (HAL_ADC_ConfigChannel(&hadc, &sConfig) != HAL_OK)
  //{
  //  while(1);
  //}

    /**Configure for the selected ADC regular channel to be converted. 
    */
  sConfig.Channel = ADC_CHANNEL_1;
  if (HAL_ADC_ConfigChannel(&hadc, &sConfig) != HAL_OK)
  {
    while(1);
  }
}

void HAL_ADC_MspInit(ADC_HandleTypeDef* hadc)
{
  GPIO_InitTypeDef          GPIO_InitStruct;
  
  /*##-1- Enable peripherals and GPIO Clocks #################################*/
  /* ADC3 Periph clock enable */
  ADCx_CLK_ENABLE();
  /* Enable GPIO clock ****************************************/
  ADCx_CHANNEL_GPIO_CLK_ENABLE();
  
  /*##-2- Configure peripheral GPIO ##########################################*/ 
  /* ADC3 Channel8 GPIO pin configuration */
  GPIO_InitStruct.Pin = ADCx_CHANNEL_PIN;
  GPIO_InitStruct.Mode = GPIO_MODE_ANALOG;
  GPIO_InitStruct.Pull = GPIO_NOPULL;
  HAL_GPIO_Init(ADCx_CHANNEL_GPIO_PORT, &GPIO_InitStruct);
}

void HAL_ADC_MspDeInit(ADC_HandleTypeDef *hadc)
{
  
  /*##-1- Reset peripherals ##################################################*/
  //ADCx_FORCE_RESET();
  //ADCx_RELEASE_RESET();

  /*##-2- Disable peripherals and GPIO Clocks ################################*/
  /* De-initialize the ADC3 Channel8 GPIO pin */
  HAL_GPIO_DeInit(ADCx_CHANNEL_GPIO_PORT, ADCx_CHANNEL_PIN);
}
```

Only 3 lines use Arduino syntax.
``` c++
void setup() {
Serial.begin(115200);
}
void loop() {
Serial.printf("A1:  %d", analogRead(A1));
delay(100);
}
```

Thanks, I'll try again.  I prefer to use that analogRead(...)  is simplest and most clearly.

csnol, 2017-08-27:
Tested, It is running OK on 1 channel. 
I got same value on every channel once read multi-analogport, and the value only gotten from A0 whatever others changed. 
I think it just is a single channel ADC.

``` c++

A0 =  0
A1 =  11
A0 =  0
A1 =  11
A0 =  5
A1 =  7
A0 =  8
A1 =  0
A0 =  0
A1 =  6
A0 =  11
A1 =  0
A0 =  1023
A1 =  1023
A0 =  1023
A1 =  1023
A0 =  1023
A1 =  1023
A0 =  1023
A1 =  1023
A0 =  1023
A1 =  1023
```
Thanks for your time.

Best regards.

C.S Nol

danieleff, 2017-08-27:
I added channel deselection to the end of `analogRead()` for these chips.
Now multiple pins should be usable.

csnol, 2017-08-28:
Cool!
Everything is good on ADC!
Thanks Daniel.

closed 2017-08-28 by danieleff