danieleff/STM32GENERIC #36: F4 HAL does not allow overriding of VECT_TAB_OFFSET

rogerclarkmelbourne, 2017-08-28:
I'm not sure the best way to handle this, but the F4 does not allow VECT_TAB_OFFSET to be overridden 

https://github.com/danieleff/STM32GENERIC/blob/master/STM32/system/STM32F4/HAL_Src/system_stm32f4xx.c#L111

Unlike the F1, which has a "guard" indef
https://github.com/danieleff/STM32GENERIC/blob/master/STM32/system/STM32F1/HAL_Src/system_stm32f1xx.c#L113-L116

Is the best fix for this, simply for me to manually update file , to add the #ifdef and generate a PR for this update ?

danieleff, 2017-08-28:
Yes, `#ifndef VECT_TAB_OFFSET` is fine. 
Originally the F1 did not have it either.

rogerclarkmelbourne, 2017-08-28:
OK
I'll fork your repo and do a PR

BTW. I'm working on a F4 bootloader with @victorpv .(well Victor has done most of the work)
Hence why I need to change the VECT_TAB_OFFSET

I've managed to make it work, by creating a new upload type in your boards.txt etc 

But at the moment I have to manually issue the dfu command to upload the compiled sketch

andrethomas, 2018-01-20:
I think I just did a duplicate post on https://github.com/danieleff/STM32GENERIC/issues/54

Did you get it working roger?