danieleff/STM32GENERIC #57: Serial.available() not working properly

VitorBoss, 2018-03-29:
Looking on SerialUART.cpp the function is returning a bool condition instead os number of available bytes, a lot of libs use that like Nexion.
This is the current code:
```
int HardwareSerial::available() {
    return rxEnd != rxStart;
}

```
This is the fix:
```
int HardwareSerial::available() {
    return (rxEnd != rxStart) ? rxEnd - rxStart : 0;
}

```

VitorBoss, 2018-05-14:
Updated code, now working with Nextion

Tjeerdie, 2018-07-25:
Hi, The current version does not have this fix yet. Manually adding the this code to the core works fine! I had problems with a project and this little thing. Can you please add the fix to the master branch?

VitorBoss, 2018-07-26:
int HardwareSerial::available() {
if (rxEnd >= rxStart) return (rxEnd - rxStart);
return BUFFER_SIZE + rxEnd - rxStart;
}

That work better