danieleff/STM32GENERIC #68: STM32F407 Specs

masoudr, 2018-06-21:
Hi,
First I need to say thanks for this library. I think for STM32F4 familly it is the best arduino framework in comparison with other libraries.
As you may know The STM32F407VET6 chip has 192KB of ram but when I compile my code I see 128K.
Here is the output of arduino application:
> Sketch uses 17748 bytes (3%) of program storage space. Maximum is 524288 bytes.
> Global variables use 3920 bytes (2%) of dynamic memory, leaving 127152 bytes for local variables. Maximum is 131072 bytes.

I found the port to Atom application from [here](http://stm32duino.com/viewtopic.php?t=2066). And I think it shows the true value:

> Building .pioenvs\black_F407VE\firmware.bin
> Memory Usage -> http://bit.ly/pio-memory-usage
> DATA:    [          ]   2.0% (used 4016 bytes from 196608 bytes)
> PROGRAM: [          ]   3.7% (used 19636 bytes from 524288 bytes)
>  [SUCCESS] Took 3.62 seconds
> 
Is there any diffrence of STM cores between these two apps?
Is there any limits or it is a simple faulty compiler message?
Thanks.

ChrisMicro, 2018-06-22:
Did you choose the correct MCU variant in the menu?
As far as I know the memory definitions in the linker file are responsible. It seems that the RAM is set to 128K there:
https://github.com/danieleff/STM32GENERIC/blob/master/STM32/variants/BLACK_F407VE/ldscript.ld

masoudr, 2018-06-22:
Thanks. The only correct option is "Black F407VE (V2.0)". I edited the `ldscript.ld` and changed the RAM value. But It still shows the same message and FYI the Atom port uses the exact same files so `ldscript.ld` is same. I think the problem is somewhere else maybe compiler is the problem.

burner-, 2020-07-20:
Looks that his is not so straight forward thing. There is article what explain why it is 128kB instead on 192kB https://jeelabs.org/2018/z80-zexall-f407/