danieleff/STM32GENERIC #74: I2S library some issues

madias123, 2018-08-30:
Hello,
I hope you do not lost interest in further development this core!
There are several issues with the I2S library, tested on a STM32F407VET (black) board.
First, a bug: 
Using two instances of the library I2S + I2S2 (=SPI2+SPI3) causes a crash with the I2S.write(val); function. 
To be clear: Both (I2S,I2S2) are working in single mode without problems.
But using:
`I2SClass I2S_a(SPI2, PB15 /*DIN*/ , PB12 /*LRC*/, PB13 /*SCLK*/, PC6 /*MCK*/);
 I2SClass I2S(SPI3, PB5 /*DIN*/ , PA4 /*LRC*/, PB3 /*SCLK*/, PC7 /*MCK*/);`
and
` I2S.begin(I2S_PHILIPS_MODE, SAMPLINGFREQUENCY, 16);
  I2S_a.begin(I2S_PHILIPS_MODE, SAMPLINGFREQUENCY, 16);`
works until:
`I2S.write(val); ` or `I2S_a.write(val); ` is executed.
I thing the problem has something to do with the DMA function. As I know both (SPI2+3) are on the same DMA1 channel (0):
Stream4: SPI2_TX
Stream5: SPI3_TX
So maybe there must be a DMA release or something?
I suspect, this bug also occurs using I2S (SPI2) and common SPI3 both in DMA mode (or vice versa)
(I don't know if it's overall possible using two streams on the same channel simultaneously, I used only F1xx in the past and the DMA controller is different) 

Second: 
Overall functionality - Maybe it would be better using a circular buffer and a IRQ function (half buffer, full buffer) as SteveStrong used it in the leaflabs core. So the buffer will be filled automatically without taking care about when the buffer is empty. Even using half/full buffer will significant prevents dropouts. More advantage about this: This function/IRQ can set to any priority the user needs. 
More about this here:
http://stm32duino.com/viewtopic.php?f=51&t=4054

Thanks and regards
Matthias  (madias in STM32duino forum)

danieleff, 2018-09-03:
I tried the following code:

```c++
#include "I2S.h"

I2SClass I2S_a(SPI2, PB15 /*DIN*/ , PB12 /*LRC*/, PB13 /*SCLK*/, PC6 /*MCK*/);
I2SClass I2S(SPI3, PB5 /*DIN*/ , PA4 /*LRC*/, PB3 /*SCLK*/, PC7 /*MCK*/);

#define SAMPLINGFREQUENCY 44100

void setup() {
  I2S.begin(I2S_PHILIPS_MODE, SAMPLINGFREQUENCY, 16);
  I2S_a.begin(I2S_PHILIPS_MODE, SAMPLINGFREQUENCY, 16);

  I2S.write(1);
  I2S_a.write(2);

  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  
  I2S.write(1);
  I2S_a.write(2);
  
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
  delay(100);
}
```

and the STM32F407VE is blinking (although i did not probe the pins). Can you write a minimal example where it crashes?

danieleff, 2018-09-03:
I see it now, it happens on the first buffer send, after the first 2k samples
```c++
for (int i = 0; i < 2047 + 1; i++) {
  I2S.write(1);
  I2S_a.write(2);
}
```

madias123, 2018-09-03:
Ok, it happend earlier to me, because I set up the buffer to 64 because of the latency. After 2k: This is exactly the size of the DMA buffer (2048), so after the first time the buffer is full it hangs using two instances.

madias123, 2018-09-08:
I posted a "generic" I2S code with DMA circular buffer (full / half full IRQ) in the forum. Maybe this is the better approach for a library:
http://www.stm32duino.com/viewtopic.php?f=51&t=4095